poppy-project/poppy-ergo-jr #23: le bloc "get .......... of motors ....." ne fonctionne pas avec le bloc "all motors"

show0k, 2016-02-02:


show0k, 2016-02-02:
J'ai testé aussi avec un ancien fichier demo (où a fonctionnait avant) et ça ne fonctionne plus. 

peut-être que le problème ne vient pas des blocs mais d'autre chose ?

[--- Commented from Asana.com
#commenter Stephanie Noirpoudre
---[aa]

show0k, 2016-02-02:
voir le sprite "poule"
avant ça fonctionnait

[--- Commented from Asana.com
#commenter Stephanie Noirpoudre
---[aa]

show0k, 2016-02-07:
C'est simplement que je n'avais pas pushé la modif que j'avais fait. Encore un fois (oui, tu m'avais déjà posé cette question) s'il n'y a pas motor(s) c'est que ce n'était pas prévu d'avoir plusieurs moteurs. Il n'y a donc rien de cassé, juste une update que tu m'avais demandé qui n'a jaméis été envoyé dans la release officielle

[--- Commented from Asana.com
#commenter Théo Segonds
---[aa]

pierre-rouanet, 2016-02-25:
Status update?

pierre-rouanet, 2016-05-17:
Status update plz :)

show0k, 2016-05-17:
Fixed https://github.com/poppy-project/pypot/blob/master/changelog.md#snap

closed 2016-05-17 by show0k