poppy-project/poppy-ergo-jr #36: Select timeout error --> Segmentation fault

matthieu-lapeyre, 2016-04-19:
We have a strange issue with a standard Ergo Jr kit (Pixl, rapsberry pi, pypot v2.11.1).

The robot instanciation raises this error:

```Python
Python 2.7.10 |Continuum Analytics, Inc.| (default, Oct 28 2015, 19:48:38) 
Type "copyright", "credits" or "license" for more information.

IPython 4.0.0 An enhanced Interactive Python.

In [1]: import poppy_ergo_jr

In [2]: poppy = poppy_ergo_jr.PoppyErgoJr()
select timeout
select timeout
Segmentation fault
```

Communication with motors seems ok:
``` Python
import pypot.dynamixel
dxl = pypot.dynamixel.Dxl320IO('/dev/ttyAMA0')
dxl.scan(range(10))
>> [1, 2, 3, 4, 5, 6]
```

But not the camera ...
```
poppy@poppy:~ $ raspistill -o cam.jpg
mmal: No data received from sensor. Check all connections, including the Sunny one on the camera board
```

matthieu-lapeyre, 2016-04-19:
Ok so same setup with a new camera does not raise the error and the ergo seems to work fine.

If it is indeed a problem with broken or nearly broken camera, maybe should we add more explicit error and start the ergo with the camera disabled ?

pierre-rouanet, 2016-04-20:
Explanation here: https://www.raspberrypi.org/forums/viewtopic.php?t=35689&p=300710

The problem is that this segfault is not so easy to try/catch. I think we should at least explain this in the FAQ.

I'll try to see if we can do better.

lpierron, 2016-12-09:
I have the same problem now. The camera worked fine this morning and stop working, now I have the "select timeout" error.
I tried the explanations from https://www.raspberrypi.org/forums/viewtopic.php?t=35689&p=300710 but I do not have the same system module, I am not sure what can I do.
I reinitialize the camera raspi-config, seemed to sound OK but always 'select timeout' after reboot.
The camera red light lights for some seconds and stops.

Help please

pierre-rouanet, 2016-12-12:
Hi @lpierron,

You should check if the cable is still plugged correctly. We also had a lot of issue with camera burning :(

Unfortunately from a software point of view, there is not much that can be done. It's directly the camera driver that crashes.

lpierron, 2016-12-13:
Thanks  @pierre-rouanet the cable was unplugged !

closed 2016-12-14 by pierre-rouanet