poppy-project/poppy-ergo-jr #49: Ergo Jr mouvements lag on RPi 3

matthieu-lapeyre, 2016-09-28:
Tests done after a fresh install of the last release image

Here are two videos to compare RPi 2 & 3 with the same dance primitive, the same robot, the same SD card, connected by ethernet.
Rpi2: https://www.youtube.com/watch?v=VN0-8D-YalY
RPi3 : https://www.youtube.com/watch?v=C2j_SZYeXww

For an unknown reason, it's even worse if the RPi3 is connected by wifi.

show0k, 2016-09-29:
Eureka ! The serial used in rpi3 is a software serial provided by miniuart. The hardware UART is used for the Bluetooth...
It is super simple to switch which interface use miniuart thanks to a linux overlay. You have to add `dtoverlay=pi3-miniuart-bt` in /boot/config.txt and change the configuration of the robot to use /dev/ttyAMA0 again ; It will works as well as for rpi2 after a reboot.

Note that the Bluetooth will still work but with poor performances (and heavy CPU load).
We will update the poppy-ergo-jr image soon.

closed 2016-09-29 by show0k

matthieu-lapeyre, 2016-09-29:
It seems to work better :)

Thanks @show0k