poppy-project/poppy-ergo-jr #60: URDF model

rstrudel, 2019-11-28:
Hello,

First, thanks for open-sourcing the robots and platform in general! I am using [Pinocchio ](https://github.com/stack-of-tasks/pinocchio) and trying to plug the Poppy Ergo Jr robot inside. However, I can see that you updated the [stl files](https://github.com/poppy-project/poppy-ergo-jr/tree/master/hardware/STL) in the master branch without updating the [urdf file](https://github.com/poppy-project/poppy-ergo-jr/blob/master/software/poppy_ergo_jr/poppy_ergo_jr.urdf). The urdf points to the stl files of the URDF branch and as is the master branch does not work for me. Would it be possible to update the urdf file of the master branch ?

I would like to use the Poppy Ergo Jr robot with an horizontal gripper (the default urdf is with the gripper in vertical mode). However the stl files of the URDF branch are in 5 sections only whereas I can see the stl of the master branch are break down into more pieces. Thus if I had the urdf of the master branch, I could modify it easily to get what I want.

Thank you,

Robin

ymollard, 2020-06-11:
Hello @rstrudel, I am currently working on an updated version of the URDF of the Poppy Ergo Jr [there](https://github.com/poppy-project/poppy_ergo_jr_description)
It currently only works with the gripper, although I'd like to adapt it for the lamp and the pen holder with xacro.

rstrudel, 2020-06-11:
Hello @ymollard, great thanks for the notice!
I created a [repository](https://github.com/rstrudel/poppyarm) where I interface the robot with pinocchio, I am not actively working on the Poppy robot as of now but I may update it with your description at some point.